eclipse-milo/milo #1113: Artifact Ids should have "milo" in them, to be more distinct and self-contained

in-fke, 2023-04-04:
Looking at the S-BOM, 

```
org.eclipse.milo:bsd-core:jar:0.6.8:compile
org.eclipse.milo:bsd-generator:jar:0.6.8:compile
org.eclipse.milo:sdk-client:jar:0.6.8:compile
org.eclipse.milo:sdk-core:jar:0.6.8:compile
org.eclipse.milo:sdk-server:jar:0.6.8:compile
org.eclipse.milo:stack-client:jar:0.6.8:compile
org.eclipse.milo:stack-core:jar:0.6.8:compile
org.eclipse.milo:stack-server:jar:0.6.8:compile
```

I find that the resulting jar files, e.g. "sdk-client-0.6.8.jar", are not very distinct.
e.g. https://mvnrepository.com/search?q=sdk-client

Maybe the Artifact Ids should all be prefixed with "milo-" ? Is there a common naming policy for Eclipse Projects with this regard?

kevinherron, 2023-04-04:
I don't know if there is a common naming policy.

I've thought about these artifact names in the past, but never really been able to convince myself it's an issue one way or the other. They are unambiguous when using the fully qualified Maven coordinate. What does it really matter what the JAR file is called?

kevinherron, 2023-04-04:
Also, you can easily resolve this in your own project with the `prependGroupId` configuration parameter for `maven-dependency-plugin`. 

I imagine you could also arbitrarily rename dependencies as they are copied somehow, and that there's an equivalent for Gradle.

in-fke, 2023-04-05:
Thanks for quick follow-up, Kevin.

Take this group as an example:
https://mvnrepository.com/artifact/org.springframework
all artifacts start with "spring-".

`prependGroupId` would only be relevant for `maven-dependency-plugin`, but not for copying to WEB-INF/lib etc. and it would be a "hassle" for integrators to change the namings in this manner.

You don't need to hurry with this decision, but my personal view is that most projects have "self-contained" prefixes in their artifactId, e.g. here it could be "milo-".

kevinherron, 2023-04-05:
I'll consider this change for the eventual 1.0 release.

kevinherron, 2023-04-06:
Implemented in `dev/1.0` branch.

commit b6da00a6 (2023-04-06): Add "milo-" prefix to Maven artifact ids

fixes #1113
closed 2023-04-06 by kevinherron