eclipse-milo/milo #1114: Client connect() never returns, stuck after receiving OpenSecureChannelResponse

milgner, 2023-04-05:
**Describe the bug**

After setting up a client with an `EndpointDescription`, I call `opcuaClient.connect().get()` in order to initiate the connection.

In Wireshark I can see that the expected UACP HEL/ACK messages are exchanged and the subsequent `OpenSecureChannelRequest` and `OpenSecureChannelResponse` messages. The `StatusCode` of the response is `GOOD` and it contains a valid SecurityToken. 

However, in my client application, nothing happens afterwards and `connect().get()` doesn't return. Since the examples don't show any code that explicitly establishes a session, I'd have expected a `CreateSessionRequest` for the endpoint config to be sent out next.

```
private val opcuaClient by lazy {
        val endpoint = EndpointDescription.builder()
            .endpointUrl("opc.tcp://localhost:4840/UA")
            .securityMode(MessageSecurityMode.None)
            .securityPolicyUri(SecurityPolicy.None.uri)
            .transportProfileUri(TransportProfile.TCP_UASC_UABINARY.uri)
            .securityLevel(ubyte(0))
            .build()
        val config = OpcUaClientConfigBuilder()
            .setEndpoint(endpoint)
             // tried with and without this code from the examples, did not change anything (these are only used in the createSessionRequest anyway, afaik
            .setKeyPair(keyStoreLoader.clientKeyPair)
            .setCertificate(keyStoreLoader.clientCertificate)
            .setCertificateChain(keyStoreLoader.clientCertificateChain)
            .setSessionTimeout(uint(120000))
            .setRequestTimeout(uint(2000))
            .setConnectTimeout(uint(2000))
            .setIdentityProvider(AnonymousProvider())
            .setCertificateValidator(ClientCertificateValidator.InsecureValidator())
            .build()
        OpcUaClient.create(config)
    }

   override suspend fun start() {
        opcuaClient.connect().await()
  
      // we never get here
        opcuaClient.session.await()
  }
```

[connect.pcapng.gz](https://github.com/eclipse/milo/files/11158117/connect.pcapng.gz)

**Expected behavior**

The `CreateSessionRequest` is sent out to enable calling the Read service afterwards.

**Additional context**

Using Kotlin Coroutines with Vert.x framework. 
Current Milo version.
Server is a defects inspection system which is apparently implement with the Unified Automation C++ SDK.

Other code using `CompleteableFuture` works fine, i.e. Hibernate Reactive.

kevinherron, 2023-04-05:
If you set the logging level for `org.eclipse.milo.opcua.sdk.client.SessionFsm` to DEBUG there should be some additional logging that may help figure out what's happening.

kevinherron, 2023-04-05:
Actually, never mind, I can see what's wrong in the capture.

The OpenSecureChannelRequest has SequenceNumber=1:
[image]

The OpenSecureChannelResponse comes back with SequenceNumber=51, and doesn't get matched up:
[image]

Seems like a bug in that server.

kevinherron, 2023-04-05:
Ah, I suspect that you're self-constructed EndpointDescription, which is missing the `Server` field (an `ApplicationDescription`), is causing an NPE inside the session FSM:
https://github.com/eclipse/milo/blob/3235f91e034d830a409039543113b5d00f626bf5/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/session/SessionFsmFactory.java#L806-L815

milgner, 2023-04-05:
Wow, thanks for the quick feedback! Unfortunately the hostname in the endpoint list returned by the server is incorrect, so I can't use the regular handshake mechanism. (it often is as few industry networks have DNS and when there's DNS it often gets torpedoed by containers or NAT.)

But I have added the line 

```
            .server(ApplicationDescription.builder().build())
```

now and was able to issue the `CreateSessionRequest`, and the corresponding response header has status code `Good`, too. 
The app now complains "no anonymous token policy found" - but I'm confident that I'll be able to pick it up from here.

I guess I was mostly missing the logs - I hadn't yet configured the slf4j to log4j2 bridge but after adding that, too, I can finally see what's going on.

closed 2023-04-05 by milgner

kevinherron, 2023-04-05:
> Unfortunately the hostname in the endpoint list returned by the server is incorrect, so I can't use the regular handshake mechanism.

You can. Once you get the `EndpointDescription` back, use `EndpointUtil.updateUrl` to correct the hostname (and port) if necessary. https://github.com/eclipse/milo/blob/18b716c0f87b8431a5ab9ae734ecf6d1d0e71338/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/util/EndpointUtil.java#L92-L108

This is actually a very common problem that clients should be able to handle.