eclipse-milo/milo #1130: OPC UA 1.05 and Milo 1.0.0 Release

kevinherron, 2023-06-12:
Parent ticket for all 1.0 sub-tasks and issues.

kevinherron, 2023-06-12:
ETA... late 2023, early 2024. Definitely not before the Germany OPC interop event in October.

kevinherron, 2024-01-22:
I don't know, I'm about to resume working on it. There's a chance it could be ready by April, after I've attended the OPC interoperability event in March.

The time I thought was going to be allocated to working on it at my job vs. on the side got reallocated to something higher priority.

apupier, 2024-04-19:
provided https://github.com/eclipse/milo/pull/1263 for the migration from testNG to JUnit 5

andvasp, 2024-06-28:
Hi @kevinherron,

Since there are breaking changes in this version (specially  #1223), would it not be good to have a version 0.7 based on 1.0?

Thanks!

kevinherron, 2024-06-28:
I don't see the point in that.

andvasp, 2024-06-28:
The point is that with a published version it will be easier to use (migrate our code), test and even suggest any modifications to the API prior to 1.0, which should be more stable. And you don't need to worry about all the requirements to release version 1.0. 

A lot of other systems can enjoy this version too as mentioned here #1098.

Why not incremental releases?

Anyway thanks for the great job you are doing here!

kevinherron, 2024-06-28:
You can develop against `dev/1.0`.

I don't want to do an intermediate release with a bunch of breaking changes, when it's a certainty that 1.0 will introduce more. It's just extra work I'd be asking people to deal with. 1.0 will be one big ugly migration effort which you can start at any time you like by developing against `dev/1.0`. 

When 1.0 is released I have no intention of supporting any 0.x releases, which will be another pain point for anyone who migrated to a hypothetical 0.7.x release.

kevinherron, 2024-06-28:
Sorry for the gruff response, I've got a lot on my plate between now and the end of the year 😬

andvasp, 2024-07-01:
Don't worry. I didn't consider it as a rugged response.

stegal-bh, 2024-09-11:
Hi guys, any update about ETA for the 1.0 release ?

kevinherron, 2024-09-11:
Nobody but me occasionally working on this. Maybe something in January? Who knows.

Anybody who has time to build and test with `dev/1.0` branch would be helping me out.

andvasp, 2024-09-13:
Hi @kevinherron!

I will try to test it.

kcello, 2024-09-18:
Hi @kevinherron,

I'm currently using org.eclipse.milo version 0.5.4 in a Spring Boot application that's compiled with GraalVM for a native Windows application. I would like to upgrade to version 1.0 to take advantage of the latest features and improvements.

I have a couple of questions:

Which dependency version should I specify in my pom.xml to use the dev/1 (version 1.0) branch of org.eclipse.milo?
Which repository should I add to my pom.xml to obtain this version?
If the dev/1 version isn't available in any public Maven repositories, could someone guide me on how to compile it from the source code and what I need to add to my pom.xml to include it in my project?

hyslopc, 2024-12-01:
@kevinherron FYI we moved our client from 0.6.14 to 1.0.0 without encountering any major issues.  I definitely prefer the 1.0 API.

kevinherron, 2024-12-01:
> @kevinherron FYI we moved our client from 0.6.14 to 1.0.0 without encountering any major issues. I definitely prefer the 1.0 API.

@hyslopc thanks for the feedback! I know it's not a drop-in replacement, but I tried to use the last 5+ years of supporting the 0.x APIs to come up with something that is easier to figure out and still supports most/all use cases.

johannwesely, 2024-12-18:
Hello Kevin,
I would like to ask cautiously if you can already give a more precise release date for 1.0 or if you can estimate when it will be released? Sorry to bother you but Managment ist asking stupid questions ㋡.

Greetings Johann

kevinherron, 2024-12-18:
No sorry, no idea really. The time to work on Milo during my day job has not really materialized the way I thought, project is delayed delay delayed. At this point I'm considering waiting until I can test it at the OPC UA interoperability event in March.

system24seven, 2024-12-31:
Is the initial list still current for outstanding items?

kevinherron, 2024-12-31:
> Is the initial list still current for outstanding items?

Yeah, fairly accurate. DynamicStructCodec has some work done on it. ECC will probably not make it for 1.0. Anything you’re interested in?

system24seven, 2024-12-31:
I was thinking of looking into the more mundane things like the code style and JSpecify stuff if it still needs doing?

kevinherron, 2024-12-31:
> I was thinking of looking into the more mundane things like the code style and JSpecify stuff if it still needs doing?

JSpecify and unit test conversion would be useful. 

Don’t worry about the code style, I’m actually going to adopt the Google check style rules and google-java-format util, so that tSK will be very trivial. Just recently decided on that so I’ll probably apply it soon.

kevinherron, 2024-12-31:
😳 https://github.com/eclipse-milo/milo/pull/1355

system24seven, 2024-12-31:
#1356 should have all the JSpecify updates, no new declarations. For some reason #1355 has a broken workflow in github, if I fork it, it fails to build.

kevinherron, 2024-12-31:
Thanks! Merged now.

system24seven, 2025-01-01:
#1358 is apupier's JUnit migration code but I have merged in the dev/1.0 code and updated everything as best as I can. 
There could be a few asserts that are mixed up, I went through everything relatively manually to do the changes in order to catch the edge cases. I can't see any testNG code in the project with text searches now, so thats a start.

system24seven, 2025-01-10:
I would try and tackle other parts of this TODO list, I just struggle to find context and information on what is required for the tasks mentioned.

kevinherron, 2025-01-10:
@system24seven I’ll think about some other tasks that could be done and try to write up tickets. This list isn’t exhaustive and was more of a “note to self” when I wrote it. 

I do know that only the XML decoder is implemented, and it would be nice to have the encoder for 1.0 as well.

johannwesely, 2025-02-10:
Hello Kevin,

i tried the new DynamicStructType 
and stumbled over the Problem that de definition of BINARY_ENCODING_ID in my own custom structs was pointing to something like this: `public static final ExpandedNodeId BINARY_ENCODING_ID = ExpandedNodeId.parse(String.format("nsu=%s;s=%s", ApplicationStrings.TOXCOMUA.getStr(), "\"Root\".\"Types\".\"DataTypes\".\"BaseDataType\".\"Structure\".\"XYPointStruct\".BinaryEncoding"));` now i needed to remove the .BinaryEncoding on the end otherwise the custom struct couldn't be decoded by the client.  Or do i get something wrong ? Greetings Johann

kevinherron, 2025-04-22:
@protogenes I'll revert to 4.1.119.Final today.

kevinherron, 2025-04-24:
Milo version `1.0.0-M1` has been released and is available in Maven Central.

The default branch for this repository is now `1.0`.

kevinherron, 2025-05-08:
Version `1.0.0-M2` was published on Monday.

The release review has concluded and 1.0.0 will be released ~2025-05-19 (May 19).

fivemeyestoreserver, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

ayestoro, 2025-05-10:
This is a well-structured codebase. I appreciate the clean separation of concerns.

closed 2025-05-19 by kevinherron