eclipse-milo/milo #1157: use milo opcua server demo in window

cuican6, 2023-09-07:
**Describe the bug**
run milo opcua server in window10, use ua expert connect opcuaserver, client report :

15:20:11.507 | DiscoveryWidget    |                                | Discovery FindServers on opc.tcp://127.0.0.1:12686 failed (BadTcpEndpointUrlInvalid)
15:20:11.515 | DiscoveryWidget    |                                | Discovery GetEndpoints on opc.tcp://127.0.0.1:12686 failed (BadTcpEndpointUrlInvalid)

cuican6, 2023-09-07:
this is log
15:06:49,549 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback.groovy]
15:06:49,550 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
15:06:49,551 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [file:/E:/java/OPCUA/milo-master/milo-examples/server-examples/target/classes/logback.xml]
15:06:49,628 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
15:06:49,654 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.ConsoleAppender]
15:06:49,658 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [STDOUT]
15:06:49,737 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - This appender no longer admits a layout as a sub-component, set an encoder instead.
15:06:49,737 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - To ensure compatibility, wrapping your layout in LayoutWrappingEncoder.
15:06:49,737 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - See also http://logback.qos.ch/codes.html#layoutInsteadOfEncoder for details
15:06:49,738 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to INFO
15:06:49,738 |-INFO in ch.qos.logback.core.joran.action.AppenderRefAction - Attaching appender named [STDOUT] to Logger[ROOT]
15:06:49,738 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
15:06:49,739 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@13becad5 - Registering current configuration as safe fallback point

15:06:49.744 [milo-nonce-util-secure-random] INFO  o.e.m.o.stack.core.util.NonceUtil - SecureRandom seeded in 51ms.
15:06:49.749 [main] INFO  o.e.m.examples.server.ExampleServer - security dir: C:\Users\Mr\AppData\Local\Temp\server\security
15:06:49.749 [main] INFO  o.e.m.examples.server.ExampleServer - security pki dir: C:\Users\Mr\AppData\Local\Temp\server\security\pki
15:06:49.754 [main] INFO  o.e.m.examples.server.KeyStoreLoader - Loading KeyStore at C:\Users\Mr\AppData\Local\Temp\server\security\example-server.pfx
15:06:50.696 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Java version: 17.0.7
15:06:50.697 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Stack version: 0.6.10
15:06:50.697 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Server SDK version: 0.6.10
15:06:52.249 [main] INFO  o.e.m.o.s.s.n.OpcUaNamespace - Loaded 2004 nodes in 738ms.
15:06:52.549 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://DESKTOP-V9CS5B0:12686/milo to 0.0.0.0:12686 [None/None]
15:06:52.959 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://DESKTOP-V9CS5B0:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
15:06:52.959 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://DESKTOP-V9CS5B0:12686/milo/discovery to 0.0.0.0:12686 [None/None]
15:06:52.959 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.0.2:12686/milo to 0.0.0.0:12686 [None/None]
15:06:52.959 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.0.2:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://192.168.0.2:12686/milo/discovery to 0.0.0.0:12686 [None/None]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://activate.navicat.com:12686/milo to 0.0.0.0:12686 [None/None]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://activate.navicat.com:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://activate.navicat.com:12686/milo/discovery to 0.0.0.0:12686 [None/None]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo to 0.0.0.0:12686 [None/None]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://127.0.0.1:12686/milo/discovery to 0.0.0.0:12686 [None/None]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://DESKTOP-V9CS5B0:8443/milo to 0.0.0.0:8443 [None/None]
15:06:52.960 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://DESKTOP-V9CS5B0:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
15:06:52.961 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://DESKTOP-V9CS5B0:8443/milo/discovery to 0.0.0.0:8443 [None/None]
15:06:52.961 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://192.168.0.2:8443/milo to 0.0.0.0:8443 [None/None]
15:06:52.961 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://192.168.0.2:8443/milo to 0.0.0.0:8443 [Basic256Sha256/Sign]
15:06:52.961 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint https://192.168.0.2:8443/milo/discovery to 0.0.0.0:8443 [None/None]
[... cut ...]
Caused by: org.eclipse.milo.opcua.stack.core.UaException: unrecognized endpoint url: opc.tcp://127.0.0.1:12686
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerHelloHandler.onHello(UascServerHelloHandler.java:140)
	at org.eclipse.milo.opcua.stack.server.transport.uasc.UascServerHelloHandler.decode(UascServerHelloHandler.java:110)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:510)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:449)
	... 16 common frames omitted

cuican6, 2023-09-07:
sorry, my code is err

closed 2023-09-07 by cuican6

kevinherron, 2023-09-07:
The endpoint URLs used by the example server are:
- opc.tcp://127.0.0.1:12686/milo 
- opc.tcp://127.0.0.1:12686/milo/discovery

This does not include:
- opc.tcp://127.0.0.1:12686