eclipse-milo/milo #1163: opcua monitors disconnection attempts several times?

1198735966, 2023-09-18:
opcua monitors disconnection attempts several times?

1198735966, 2023-09-18:
private class CustomSubscriptionListener implements UaSubscriptionManager.SubscriptionListener {

        public void onKeepAlive(UaSubscription subscription, DateTime publishTime) {
            UaSubscriptionManager.SubscriptionListener.super.onKeepAlive(subscription, publishTime);
            LogUtil.writelog("opcUa监听：onKeepAlive（连接存活中）", logName);
        }

        public void onStatusChanged(UaSubscription subscription, StatusCode status) {
            UaSubscriptionManager.SubscriptionListener.super.onStatusChanged(subscription, status);
            LogUtil.writelog("opcUa监听：onStatusChanged（连接状态改变）:" + status, logName);
        }

        public void onPublishFailure(UaException exception) {
            UaSubscriptionManager.SubscriptionListener.super.onPublishFailure(exception);
            boolean cancelled = opcLink.getSession().isCancelled();
            LogUtil.writelog("opcUa监听：onPublishFailure（连接断开）", logName);

//            //连接断开记录
//            OpcUaConfig.opcuaRecord();
        }

        public void onNotificationDataLost(UaSubscription subscription) {
            UaSubscriptionManager.SubscriptionListener.super.onNotificationDataLost(subscription);
            LogUtil.writelog("opcUa监听：onNotificationDataLost（数据丢失）:" + subscription, logName);
        }

        /**
         * 重连时 尝试恢复之前的订阅失败时 会调用此方法
         *
         * @param uaSubscription 订阅
         * @param statusCode     状态
         */
        public void onSubscriptionTransferFailed(UaSubscription uaSubscription, StatusCode statusCode) {
            UaSubscriptionManager.SubscriptionListener.super.onSubscriptionTransferFailed(uaSubscription, statusCode);
            //在回调方法中重新订阅
            LogUtil.writelog("opcUa监听：正在重新启动订阅22", logName);
            // 获取OPC UA服务器  --参数true就是是否需要重复连接
//            createSubscription(true);
        //   opcLink.getSubscriptionManager().clearSubscriptions();
//            OpcuaClient.opcLink.getSubscriptionManager().removeSubscriptionListener(this);
//            重启
            run();
        }
    }

//issues
opcua suddenly disconnects，Will trigger several onPublishFailure？Will trigger several onSubscriptionTransferFailed？

kevinherron, 2023-09-18:
These are expected to fire multiple times (when appropriate):
- `onPublishFailure` will fire for each pending PublishRequest that failed
- `onSubscriptionTransferFailed` will fire after you reconnect for each subscription that could not be transferred to the new session

If you're just trying to get notified that a connection was lost or regained then use a `SessionActivityListener` instead.