eclipse-milo/milo #1199: Force TCP reconnect after SessionFsm keep-alive failure

kevinherron, 2024-01-10:
If a server goes offline in an "unclean" manner, the keep-alive kicks in and session re-activation is attempted, but the underlying TCP connection may not be detected as lost for quite some time (depending on how the OS TCP stack treats TCP keep alive).

When using the OPC TCP transport we should close the underlying channel when the keep-alive failure occurs to force a reconnect at that layer as well.

closed 2024-01-10 by kevinherron

weimengxz, 2024-05-10:
So this issue has been fixed in the latest version? And how long does it take to finish the re-connection in the latest version?

kevinherron, 2024-05-10:
Yes, it was part of the 0.6.12 release.

If a server "disappears" in an unclean manner that doesn't result in the TCP connection being broken then how long the reconnect takes is a function of the keep-alive settings you've configured your client with. By default it would be ~10 seconds.

PR #1200 (2024-01-10): Force TCP reconnect when session keep-alive failure occurs
fixes #1199