eclipse-milo/milo #1216: URI Reading Error

InformatikerImAll, 2024-01-31:
I have the following Errormessage when i start my Application. 

`2024-01-31T09:56:18.171 UTC ERROR [main|org.eclipse.milo.opcua.stack.core.util.ManifestUtil:98] uris() failed
java.net.URISyntaxException: Illegal character in opaque part at index 30: jar:nested:/C:/Users/.../energy-belvispluscontrol-provis.deployment/energy-belvispluscontrol.provis.distribution/target/energy-belvispluscontrol.provis.distribution-1.0.0-SNAPSHOT.jar/!BOOT-INF/classes/!/META-INF/MANIFEST.MF
        at java.base/java.net.URI$Parser.fail(URI.java:2974)
        at java.base/java.net.URI$Parser.checkChars(URI.java:3145)
        at java.base/java.net.URI$Parser.parse(URI.java:3181)
        at java.base/java.net.URI.<init>(URI.java:623)
        at java.base/java.net.URL.toURI(URL.java:1056)
        at org.eclipse.milo.opcua.stack.core.util.ManifestUtil.uris(ManifestUtil.java:93)
        at org.eclipse.milo.opcua.stack.core.util.ManifestUtil.load(ManifestUtil.java:46)
        at org.eclipse.milo.opcua.stack.core.util.ManifestUtil.<clinit>(ManifestUtil.java:33)
        at org.eclipse.milo.opcua.sdk.client.OpcUaClient.<clinit>(OpcUaClient.java:141)
        at ....createClient(ProcosCoreConnection.java:239)
        at ....initializeConnection(ProcosCoreConnection.java:800)
        at ....doConnect(ProcosCoreConnection.java:783)
        at ....openConnection(AbstractCoreConnection.java:42)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:568)
        at org.springframework.aop.support.AopUtils.invokeJoinpointUsingReflection(AopUtils.java:352)
        at org.springframework.aop.framework.JdkDynamicAopProxy.invoke(JdkDynamicAopProxy.java:242)
        at jdk.proxy2/jdk.proxy2.$Proxy98.openConnection(Unknown Source)
        at ....afterPropertiesSet(DelegatingProcessVariableObservable.java:35)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.invokeInitMethods(AbstractAutowireCapableBeanFactory.java:1822)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1771)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:601)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:523)
        at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:325)
        at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234)
        at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:323)
        at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:199)
        at org.springframework.beans.factory.config.DependencyDescriptor.resolveCandidate(DependencyDescriptor.java:254)
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.doResolveDependency(DefaultListableBeanFactory.java:1441)
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.resolveDependency(DefaultListableBeanFactory.java:1348)
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.resolveFieldValue(AutowiredAnnotationBeanPostProcessor.java:769)
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:752)
        at org.springframework.beans.factory.annotation.InjectionMetadata.inject(InjectionMetadata.java:145)
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor.postProcessProperties(AutowiredAnnotationBeanPostProcessor.java:493)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.populateBean(AbstractAutowireCapableBeanFactory.java:1420)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:600)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:523)
        at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:325)
        at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234)
        at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:323)
        at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:199)
        at org.springframework.beans.factory.config.DependencyDescriptor.resolveCandidate(DependencyDescriptor.java:254)
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.doResolveDependency(DefaultListableBeanFactory.java:1441)
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.resolveDependency(DefaultListableBeanFactory.java:1348)
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.resolveFieldValue(AutowiredAnnotationBeanPostProcessor.java:769)
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:752)
        at org.springframework.beans.factory.annotation.InjectionMetadata.inject(InjectionMetadata.java:145)
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor.postProcessProperties(AutowiredAnnotationBeanPostProcessor.java:493)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.populateBean(AbstractAutowireCapableBeanFactory.java:1420)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:600)
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:523)
        at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:325)
        at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234)
        at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:323)
        at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:199)
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:973)
        at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:946)
        at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:616)
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:146)
        at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:753)
        at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:455)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:323)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1342)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1331)
        at ....ApplicationStarter.main(ApplicationStarter.java:19)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[... cut ...]
        at org.springframework.boot.loader.launch.PropertiesLauncher.main(PropertiesLauncher.java:574)` Even tho it thwrows this Error the connection to the OPC-UA Server seems to work perfectly fine. I checked evry uri twice but it always throws this Message.  Here is the Code-Part that i use.             ```
OpcUaClient opcUaClient = OpcUaClient.create(
                    this.endpointUrl,
                    endpoints -> endpoints.stream().filter(this.endpointFilter()).findFirst(),
                    configBuilder -> buildConfig(configBuilder, loader, certificateValidator));
```
this.endpointUrl looks like this: opcua.endpointurl=opc.tcp://ip:port

kevinherron, 2024-01-31:
Look at the contents of the `META-INF/MANIFEST.MF` in the JAR file you build.

InformatikerImAll, 2024-02-01:
> Look at the contents of the `META-INF/MANIFEST.MF` in the JAR file you build.

Thats what is in the Manifest.mf 
`Manifest-Version: 1.0
Created-By: Maven Jar Plugin 3.2.0
Build-Jdk-Spec: 17
Main-Class: org.springframework.boot.loader.launch.PropertiesLauncher
Start-Class: de.kisters.energy.belvispluscontrol.provis.launcher.Applica
 tionStarter
Spring-Boot-Version: 3.2.0
Spring-Boot-Classes: BOOT-INF/classes/
Spring-Boot-Lib: BOOT-INF/lib/
Spring-Boot-Classpath-Index: BOOT-INF/classpath.idx
Spring-Boot-Layers-Index: BOOT-INF/layers.idx

`

kevinherron, 2024-02-01:
Hmm, well this is strange. It seems, as far as I can tell, that at runtime at least one of the paths to a MANIFEST.MF ends up being an invalid URI and fails here: https://github.com/eclipse/milo/blob/58068120ee6bfa2a4e5b80cf73b9b35e84aa6a3e/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/util/ManifestUtil.java#L93

I can try to more narrowly catch this exception.

In the meantime you can just ignore it. If you are able to modify logging configuration for your application you can turn the log level to `OFF` for `org.eclipse.milo.opcua.stack.core.util.ManifestUtil`.

InformatikerImAll, 2024-02-05:
> Hmm, well this is strange. It seems, as far as I can tell, that at runtime at least one of the paths to a MANIFEST.MF ends up being an invalid URI and fails here:

Thanks for your reply. I know what is the problem now.
The URI that i used contained a space character. That caused the Exception. But like i said before everything still works fine. I will ignore it now.

kevinherron, 2024-02-06:
@InformatikerImAll if possible, can you try the build from #1218 and see if you still get an error?

InformatikerImAll, 2024-02-08:
> @InformatikerImAll if possible, can you try the build from #1218 and see if you still get an error?

Yes sure. How can i test that branch? I am using maven            

```
 <dependency>
                <groupId>org.eclipse.milo</groupId>
                <artifactId>sdk-client</artifactId>
                <version>${sdk-client.version}</version>
 </dependency>
```

kevinherron, 2024-02-08:
You would need to check out this branch locally, build and install a Milo dev version via `mvn clean install`, then update your version to point at `0.6.13-SNAPSHOT`.

InformatikerImAll, 2024-02-09:
> You would need to check out this branch locally, build and install a Milo dev version via `mvn clean install`, then update your version to point at `0.6.13-SNAPSHOT`.

The 0.6.13-SNAPSHOT has a invalid jar and pom. When i build the project i have the following error: `[ERROR] Tests run: 195, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 9.256 s <<< FAILURE! - in TestSuite
[ERROR] org.eclipse.milo.opcua.stack.core.util.validation.CertificateValidationUtilTest.testBuildAndValidate_LeafIntermediateSigned_Revoked  Time elapsed: 0.07 s  <<< FAILURE!
java.lang.AssertionError: expected [StatusCode{name=Bad_CertificateRevoked, value=0x801D0000, quality=bad}] but found [StatusCode{name=Bad_CertificateIssuerRevocationUnknown, value=0x801C0000, quality=bad}]
        at org.eclipse.milo.opcua.stack.core.util.validation.CertificateValidationUtilTest.testBuildAndValidate_LeafIntermediateSigned_Revoked(CertificateValidationUtilTest.java:232)`

So is build it with -DskipTests. But the jar and pom of version 0.6.13-SNAPSHOT are somehow invalid to request them in my project. When i try 0.6.12 the error that i mentioned in this issue still occures.

PR #1218 (2024-02-09): Refactor ManifestUtil, remove conversion to URI
fixes #1216
commit cd0f3d97 (2024-02-09): Refactor ManifestUtil, remove conversion to URI (#1218)

fixes #1216
closed 2024-02-09 by kevinherron