eclipse-milo/milo #1229: Milo 0.6.12 Gradle ambiguity error for guava 33

LocutusV0nB0rg, 2024-02-26:
**Describe the bug**
Updating my gradle dependency from `api ('org.eclipse.milo:sdk-server:0.6.11')` to `api ('org.eclipse.milo:sdk-server:0.6.12')` fails my gradle build with an ambiguity due to variants of the guava transitive dependency.

**Expected behavior**
Well... Gradle should keep building.

**Logs and Packet Captures**

```
Execution failed for task ':commons-dpi-opc:compileJava'.
> Could not resolve all files for configuration ':commons-dpi-opc:compileClasspath'.
   > Could not resolve com.google.guava:guava:33.0.0-jre.
     Required by:
         project :commons-dpi-opc > org.eclipse.milo:sdk-server:0.6.12 > org.eclipse.milo:stack-core:0.6.12
      > Cannot choose between the following variants of com.google.guava:guava:33.0.0-jre:
          - androidApiElements
          - jreApiElements
        All of them match the consumer attributes:
          - Variant 'androidApiElements' capabilities com.google.collections:google-collections:33.0.0-jre and com.google.guava:guava:33.0.0-jre:
              - Unmatched attributes:
                  - Found org.gradle.jvm.environment 'android' but wasn't required.
                  - Found org.gradle.status 'release' but wasn't required.
              - Compatible attributes:
                  - Required org.gradle.category 'library' and found compatible value 'library'.
                  - Required org.gradle.dependency.bundling 'external' and found compatible value 'external'.
                  - Required org.gradle.jvm.version '8' and found compatible value '8'.
                  - Required org.gradle.libraryelements 'classes' and found compatible value 'jar'.
                  - Required org.gradle.usage 'java-api' and found compatible value 'java-api'.
          - Variant 'jreApiElements' capabilities com.google.collections:google-collections:33.0.0-jre and com.google.guava:guava:33.0.0-jre:
              - Unmatched attributes:
                  - Found org.gradle.jvm.environment 'standard-jvm' but wasn't required.
                  - Found org.gradle.status 'release' but wasn't required.
              - Compatible attributes:
                  - Required org.gradle.category 'library' and found compatible value 'library'.
                  - Required org.gradle.dependency.bundling 'external' and found compatible value 'external'.
                  - Required org.gradle.jvm.version '8' and found compatible value '8'.
                  - Required org.gradle.libraryelements 'classes' and found compatible value 'jar'.
                  - Required org.gradle.usage 'java-api' and found compatible value 'java-api'.
```

**Additional context**
Milo Version 0.6.12 is the Version where the guava version was updated. Perhaps milo has to specify the variant explicitly?

kevinherron, 2024-02-26:
I don't know what this means. I thought I already specified the variant explicitly with the `-jre` suffix on the version.

kevinherron, 2024-02-26:
https://stackoverflow.com/questions/76897459/how-can-i-make-gradle-pick-the-right-guava-variant

LocutusV0nB0rg, 2024-02-26:
I tried overwriting the transitive depends of milo in my gradle project but had no success. However there is very little documentation of how to achieve this, so I am not suprised really...  However I think this should be solved here.

kevinherron, 2024-02-26:
This doesn't look like something that would get fixed in Milo. Looks like a gradle problem, see some of the solutions here: https://github.com/google/guava/issues/6612

Are you using a recent version of Gradle? Have you applied the `java` plugin to the project you're building this in?

closed 2024-03-01 by kevinherron