eclipse-milo/milo #1230: EndpointDescription is only chosen based on path, not hostname, preventing different certificates per-endpoint

lambdaupb, 2024-02-26:
https://github.com/eclipse/milo/blob/cd0f3d97f78de3823541229032642f8af9f00040/opc-ua-stack/stack-server/src/main/java/org/eclipse/milo/opcua/stack/server/transport/uasc/UascServerAsymmetricHandler.java#L382-L385

Matches the first EndpointDescription with requested Security and Path, but ignores the hostname im the URL.

This leads milo to choose a different endpoint than requested and then presenting a different certificate.

This is a problem when I want to keep a certificate the same for previously existing hostnames, but have an additional cert for hostnames/ips not covered by the existing certificate.

Maybe the path logic is necessary, I don't know, but EndpointDescriptions with both matching hostname and path should have precedence.

Is there another way to keep certs stable so that existing clients don't need to update their trust?
Could I differentiate Endpoints by certificate - ua clients send their requested server cert fingerprint with the request.

Or is this achieved some other way?

Thanks for the all the work on this nice project.

kevinherron, 2024-02-26:
It seems to me that the thumbprint provided by the client should be used as additional criteria in the endpoint matching logic. 

Using the thumbprint seems obviously correct to me, whereas something about trying to use the host name doesn't sit right (beyond the obvious many scenarios where a client does not provide a matching host name).

kevinherron, 2024-03-17:
Fixed in https://github.com/eclipse/milo/commit/aeb554e738e1ce201fe13267705a86907d9e65ce for 1.0 release.

PR #1235 (2024-03-17): Use thumbprint as additional criteria when matching endpoints
This allows the configuration and matching of endpoints that are otherwise identical except their certificates. This can be useful in scenarios where existing trust relationships are maintained while new certificates are introduced.

fixes #1230
commit aeb554e7 (2024-03-17): Use thumbprint as additional criteria when matching endpoints (#1235)

This allows the configuration and matching of endpoints that are otherwise identical except their certificates. This can be useful in scenarios where existing trust relationships are maintained while new certificates are introduced.

fixes #1230
closed 2024-03-17 by kevinherron