eclipse-milo/milo #1249: Cannot connect to Secured OPC Server

JaeWonyH, 2024-03-28:
I'm using eclipse milo version 0.3.6
And OPC UA Server is KEPServerEX6

This is the KEPServerEX6 Configuration.
[image]

And This is the OPC UA Client Code.

** KeyStoreLoader.java
`/*
 * Copyright (c) 2016 Kevin Herron
 *
 * All rights reserved. This program and the accompanying materials
 * are made available under the terms of the Eclipse Public License v1.0
 * and Eclipse Distribution License v1.0 which accompany this distribution.
 *
 * The Eclipse Public License is available at
 *   http://www.eclipse.org/legal/epl-v10.html
 * and the Eclipse Distribution License is available at
 *   http://www.eclipse.org/org/documents/edl-v10.html.
 */

package com.sit.stdplf.aqst.agent.opcua01.milo.util;

import java.io.File;
import java.io.FileInputStream;
import java.io.FileOutputStream;
import java.security.Key;
import java.security.KeyPair;
import java.security.KeyStore;
import java.security.PrivateKey;
import java.security.PublicKey;
import java.security.cert.X509Certificate;
import java.util.regex.Pattern;

import org.eclipse.milo.opcua.sdk.server.util.HostnameUtil;
import org.eclipse.milo.opcua.stack.core.util.SelfSignedCertificateBuilder;
import org.eclipse.milo.opcua.stack.core.util.SelfSignedCertificateGenerator;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import com.sit.stdplf.aqst.agent.opcua01.common.config.Props;
import com.sit.stdplf.aqst.agent.opcua01.common.constant.Consts;

public class KeyStoreLoader
{
    private static Logger LOGGER = LoggerFactory.getLogger ( Consts.LOGGER_NAME );

    private static final String CLIENT_ALIAS = "client-ai";
    private static final char [] PASSWORD = "password".toCharArray ();

    private X509Certificate clientCertificate;
    private KeyPair clientKeyPair;

    private static final Pattern IP_ADDR_PATTERN = Pattern.compile (
                    "^(([01]?\\d\\d?|2[0-4]\\d|25[0-5])\\.){3}([01]?\\d\\d?|2[0-4]\\d|25[0-5])$" );
    
    private final String agentId = Props.AGENT_ID;

    public KeyStoreLoader load ( final File baseDir ) throws Exception
    {
        final KeyStore keyStore = KeyStore.getInstance ( "PKCS12" );

        final File serverKeyStore = baseDir.toPath ().resolve ( "example-client.pfx" ).toFile ();
        
        //황재원
//        final File clientCertifiacteFile = baseDir.toPath ().resolve ( "93f7e6357a4a0c5fe3f5c36ef9350c5a59840806.der" ).toFile ();
//        final InputStream clientCertiInputStream = new FileInputStream ( clientCertifiacteFile );
//
//        final X509Certificate clientCerti = CertificateUtil.decodeCertificate ( clientCertiInputStream );
[... 330 lines cut ...]
            } catch ( final Exception e )
            {
                // Trace를 위한 사용자 메시지 생성
                final String errTraceMsg = LogMsgUtil.makeErrTraceMessage ( this.getClass (),
                                new Throwable ().getStackTrace () [0].getMethodName () );

                final String errUsrMsg = errTraceMsg + "Error disconnecting";
                final SPNetworkException spNetworkException = new SPNetworkException ( Consts.BUNDLE_ID, ErrCds.ERR_02_0106,
                                LangMessage.getMessage ( ErrCds.ERR_02_0106 ), errUsrMsg );

                SPExceptionHandler.exceptionLogging ( spNetworkException, LOGGER );
            }
        }
    }

}
`

If I set KEPServerEx6 securitymode to Sign. It works well. 
But If I set that to Sign and Encrypt mode, OPC Client can't connect to Server.

This is the error log.
[image]

It there any problem in my code?

JaeWonyH, 2024-03-28:
In Sign&Encrpt mode, Client don't send "CreateSessionReqeuest" to Server.

kevinherron, 2024-03-28:
Maybe a Wireshark capture would help to understand?