eclipse-milo/milo #1254: Setting a node's value to null resp. clearing the node

markuzzi, 2024-03-31:
**Describe the bug**
I am trying to clear a node's value by setting it to `null`. For this I used a `DataValue(null)` as the code suggests (see below):

```
client.writeValue(nodeId, DataValue.valueOnly(null)).get();
```

This leads to an Exception to be thrown as follows:

```
UaException: status=Bad_TypeMismatch, message=The value supplied for the attribute is not of the same type as the attribute's value.
        at org.eclipse.milo.opcua.sdk.server.util.AttributeWriter.validateDataType(AttributeWriter.java:215)
```

**Expected behavior**
I expected the node to be cleared instead.

**Additional context**
I am using milo 0.6.12.

I tried to trace down the error. What I found is:
* When sending a null value, the value is decoded into `DataValue(Variant(null))` here: https://github.com/eclipse/milo/blob/7f49b8d4cc4c1efb4a9141d2b87d1286eb3ba9e7/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/serialization/OpcUaBinaryStreamDecoder.java#L221
* The server checks for null here and throws an exception: https://github.com/eclipse/milo/blob/7f49b8d4cc4c1efb4a9141d2b87d1286eb3ba9e7/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/util/AttributeWriter.java#L215
* I think either the value should be set to `null` instead of `Variant(null)` in the first link, or it should just return `null` instead of throwing an exception in the second link.

Does that make sense?

kevinherron, 2024-03-31:
Of these two options it would change in AttributeWriter. When I wrote this I’m pretty sure the idea was that null values weren’t accepted. This should probably be an option left to the implementation to decide, but I’m not sure just changing the current behavior is a good idea.

kevinherron, 2024-03-31:
Patch for future me that allows for allowing nulls without changing existing behavior:

```
Subject: [PATCH] Allow writing null if `AllowNulls` Property is `true`
---
Index: opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/util/AttributeWriter.java
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/util/AttributeWriter.java b/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/util/AttributeWriter.java
--- a/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/util/AttributeWriter.java	(revision 7f49b8d4cc4c1efb4a9141d2b87d1286eb3ba9e7)
+++ b/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/util/AttributeWriter.java	(date 1711892046423)
@@ -23,6 +23,7 @@
 import org.eclipse.milo.opcua.sdk.server.nodes.AttributeContext;
 import org.eclipse.milo.opcua.sdk.server.nodes.UaNode;
 import org.eclipse.milo.opcua.sdk.server.nodes.UaServerNode;
+import org.eclipse.milo.opcua.sdk.server.nodes.UaVariableNode;
 import org.eclipse.milo.opcua.stack.core.AttributeId;
 import org.eclipse.milo.opcua.stack.core.Identifiers;
 import org.eclipse.milo.opcua.stack.core.StatusCodes;
@@ -119,6 +120,12 @@
         );
 
         if (attributeId == AttributeId.Value) {
+            boolean allowNulls = false;
+            if (node instanceof UaVariableNode) {
+                Boolean b = ((UaVariableNode) node).getAllowNulls();
+                allowNulls = b != null ? b : false;
+            }
+
             NodeId dataType = extract(
                 node.getAttribute(
                     internalContext,
@@ -126,7 +133,7 @@
             );
 
             if (dataType != null) {
-                value = validateDataType(context.getServer(), dataType, value);
+                value = validateDataType(context.getServer(), dataType, value, allowNulls);
             }
 
             Integer valueRank = extract(
@@ -206,13 +213,20 @@
     private static DataValue validateDataType(
         OpcUaServer server,
         NodeId dataType,
-        DataValue value) throws UaException {
+        DataValue value,
+        boolean allowNulls
+    ) throws UaException {
 
         Variant variant = value.getValue();
-        if (variant == null) return value;
 
         Object o = variant.getValue();
-        if (o == null) throw new UaException(StatusCodes.Bad_TypeMismatch);
+        if (o == null) {
+            if (allowNulls) {
+                return value;
+            } else {
+                throw new UaException(StatusCodes.Bad_TypeMismatch);
+            }
+        }
 
         Class<?> valueClass = o.getClass().isArray() ?
             ArrayUtil.getType(o) : o.getClass();
```

commit 463cd448 (2024-03-31): Allow writing `null` when "AllowNulls" Property is `true`

fixes #1254
PR #1255 (2024-03-31): Allow writing `null` when "AllowNulls" Property is `true`
fixes #1254
commit 6a67a12b (2024-03-31): Allow writing `null` when "AllowNulls" Property is `true` (#1255)

fixes #1254
closed 2024-03-31 by kevinherron