eclipse-milo/milo #1262: Misleading declaration of `CompositeValidator` compared to its use

milgner, 2024-04-19:
**Describe the bug**
`CompositeValidator` is declared as

```
public class CompositeValidator<T> implements IdentityValidator<T> {
    public T validateIdentityToken(...) {}
}
```

which implies that it can only compose validators which all return the same type `T`.

However, in the example server, it is constructed via `new CompositeValidator(identityValidator, x509IdentityValidator)` where two different types of identity validators are combined.

This results in a compilation error, at least in Kotlin (see below for workaround).

**Expected behavior**
If `CompositeValidator` should be usable with different types of validators, the declaration needs to be changed. If it should not, the example should be changed accordingly.

**Logs and Packet Captures**
N/A

**Additional context**
I'm surprised that this apparently compiles in Java? I know that the JVM has type parameter erasure at runtime but I would still have expected the Java compiler to fail as using different `T` in constructor arguments does not match the declaration.
 
When used in Kotlin, the code refuses to compile which made me fall back to this code

```
private val combinedIdentityValidator by lazy {
        val combined = listOf<IdentityValidator<*>>(usernameIdentityValidator, x509IdentityValidator)
        CompositeValidator(combined as List<IdentityValidator<Nothing>>)
    }
```

Magniveo, 2024-04-19:
```
UsernameIdentityValidator-AbstractIdentityValidator-IdentityValidator
X509IdentityValidator-AbstractX509IdentityValidator-AbstractIdentityValidator-IdentityValidator
CompositeValidator implements IdentityValidator
```
And IdentityValidator its interface. What need change?

kevinherron, 2024-04-19:
This has already been refactored in the `dev/1.0` branch.

closed 2024-04-19 by kevinherron