eclipse-milo/milo #1267: Getting Bad_SecurityChecksFailed followed by Bad_ConnectionClosed when using recursion with OpcUaClient.getAddressSpace().browseNodesAsync(UaNode).thenCompose()

ThoSap, 2024-04-30:
**Describe the bug**
Hi, I currently have the issue that I get the following error messages when using recursion with `OpcUaClient.getAddressSpace().browseNodesAsync(UaNode).thenCompose()`.
Basically, I get a
`ERROR [UascClientMessageHandler] (milo-shared-thread-pool-0) Error decoding symmetric message: org.eclipse.milo.opcua.stack.core.channel.MessageDecodeException: UaException: status=Bad_SecurityChecksFailed, message=bad sequence number: 2626, lastSequenceNumber=2566`
followed by a
WARN  [AddressSpace] (milo-shared-thread-pool-2) Failed to create Node from Reference to ExpandedNodeId{ns=2, id=BRU_ENRG_0001.10_90_129_10.BRU.BAR.S1p2.activeEnergy._Description, serverIndex=0}: java.util.concurrent.CompletionException: UaException: status=Bad_ConnectionClosed, message=connection closed

**Expected behavior**
I should not get the `MessageDecodeException` with `Bad_SecurityChecksFailed, message=bad sequence number`

**Logs and Packet Captures**
N/A

**Additional context**
Client e.g. Eclipse Milo version: 0.6.12
Server: [KEPserverEX 6.12.361.0](https://support.ptc.com/help/kepware/kepserverex/en/index.html#page/kepware/kepserverex/whats_new/whats_new_6_12_361_0.html#)

**Simplified code where this occurs**

```java
package com.example;

import org.eclipse.milo.opcua.sdk.client.OpcUaClient;
import org.eclipse.milo.opcua.sdk.client.SessionActivityListener;
import org.eclipse.milo.opcua.sdk.client.api.UaSession;
import org.eclipse.milo.opcua.sdk.client.model.nodes.objects.ServerTypeNode;
import org.eclipse.milo.opcua.sdk.client.model.nodes.variables.ServerStatusTypeNode;
import org.eclipse.milo.opcua.sdk.client.nodes.UaNode;
import org.eclipse.milo.opcua.stack.core.Identifiers;
import org.eclipse.milo.opcua.stack.core.UaException;
import org.eclipse.milo.opcua.stack.core.security.SecurityPolicy;
import org.eclipse.milo.opcua.stack.core.types.builtin.DateTime;
import org.eclipse.milo.opcua.stack.core.types.builtin.LocalizedText;
import org.eclipse.milo.opcua.stack.core.types.builtin.NodeId;
import org.eclipse.milo.opcua.stack.core.types.enumerated.NodeClass;
import org.eclipse.milo.opcua.stack.core.types.enumerated.ServerState;
import org.eclipse.milo.opcua.stack.core.types.structured.BuildInfo;
import org.eclipse.milo.opcua.stack.core.types.structured.ServerStatusDataType;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import java.util.Arrays;
import java.util.Iterator;
import java.util.List;
import java.util.concurrent.CompletableFuture;
import java.util.concurrent.CopyOnWriteArrayList;
import java.util.concurrent.ExecutionException;
import java.util.concurrent.TimeUnit;
import java.util.concurrent.TimeoutException;
import java.util.stream.Stream;

import static org.eclipse.milo.opcua.stack.core.types.builtin.unsigned.Unsigned.uint;

public class BadSecurityCheckFailedExample {

    private static final Logger log = LoggerFactory.getLogger(BadSecurityCheckFailedExample.class);

    private OpcUaClient client = null;
    private final CompletableFuture<OpcUaClient> future = new CompletableFuture<>();

    /*
     * OPC UA inactive client handling and stopping the thread
     */
    // If true, the Thread pauses as the OPC UA client is not active
    private volatile boolean stop = false;

    private synchronized boolean keepRunning() {
        return !this.stop;
    }

[... 683 lines cut ...]
           │    ├── (G) NSHV2Q19p1
           │    │    └── (V) activeEnergy
           │    ├── (G) NSHV2Q21p1
           │    │    └── (V) activeEnergy
           │    ├── (G) NSHV2Q23p1
           │    │    └── (V) activeEnergy
           │    ├── (G) NSHV2Q29p2
           │    │    └── (V) activeEnergy
           │    ├── (G) NSHV2Q30p1
           │    │    └── (V) activeEnergy
           │    ├── (G) NSHV2Q3p1
           │    │    └── (V) activeEnergy
           │    ├── (G) NSHV2Q6p2
           │    │    └── (V) activeEnergy
           │    ├── (G) NSHV2Q7p1
           │    │    └── (V) activeEnergy
           │    └── (G) NSHV2Q8p1
           │         └── (V) activeEnergy
           └── (T) MEC
                ├── (G) electricity
                │    └── (V) activeEnergy
                └── (G) gas
                     └── (V) mainCounter
2024-04-30T13:09:25.893Z INFO  [Schnurfla] (Polling connector -> Energy) Browse took 60000ms -> Found 5 nodes
```

kevinherron, 2024-04-30:
This doesn't look like a bug on the Milo side of things to me, it looks like a bug in the server, likely triggered by your unbounded recursive browse attempt.

Get a Wireshark capture of the whole thing with security disabled and we'll be able to see if the sequence numbers error is legit or not.

ThoSap, 2024-04-30:
I will provide you a Wireshark log on Thursday, I'm on PTO tomorrow.

Do you have a suggestion on how I could improve the small recursive method `BadSecurityCheckFailedExample#browseRecursive` or could you suggest a better/proper solution?

kevinherron, 2024-04-30:
Recursion is okay, it's the unbounded parallelism that becomes a problem for many servers.

ThoSap, 2024-05-02:
A similar issue I found were you found the root cause:
https://forum.inductiveautomation.com/t/anyone-know-of-a-way-to-dump-all-the-current-tag-values-from-the-opc-ua-server/74481/16

kevinherron, 2024-05-02:
Thanks, the capture and logs are perfect, I'm looking at this now.

I suspect the problem is that you sending and receiving thousands of requests/responses concurrently and triggering this logic that was added to prevent DoS attacks: https://github.com/eclipse/milo/issues/1147

You could effectively disable it and revert to previous behavior by setting the `milo.stack.serialization.maxQueueSize` property, e.g. `-Dmilo.stack.serialization.maxQueueSize= 2147483647`

ThoSap, 2024-05-02:
Unfortunately setting `-Dmilo.stack.serialization.maxQueueSize=2147483647` did not help, I get the same exceptions.
[image]

My client may be able to handle more requests as I compiled this Java `21.0.2` code AoT using GraalVM `native-image`.

The funny thing is, that I suddenly got this error without changing the Eclipse Milo version (`0.6.11`) or KEPserverEX version, even though the same code (same Docker image) ran for half a year (even with some container restarts in between K8s node maintenance -> e.g. move POD to other worker nodes).

We also did not add any new OPC UA Objects or Variables in this time frame, so literally nothing changed in this time frame until this issue suddenly appeared.
Also, this previous version using Eclipse Milo version `0.6.11` ran for months was AoT compiled using GraalVM `native-image` and Java `17.0.8.1+1`.

To try to mitigate this issue, I then updated everything (Eclipse Milo, Java, Quarkus, KEPserverEX, etc.) to see if this fixes this issue (possible bugfix in any of these components), but no luck as you can see.

ThoSap, 2024-05-02:
Now the cron schedule that runs the `BadSecurityCheckFailedExample#scrapeNodes` ran again and there I got the new error message `Could not get NodeId NodeId{ns=2, id=BRU_ENRG_0001} -> status=Bad_SecureChannelIdInvalid, description=The specified secure channel is no longer valid.`

kevinherron, 2024-05-02:
I'm pretty sure that's the issue... it's the only thing I can see that would cause it. I don't have any logging in place that can prove it, but I can add some if you're able to build and test against snapshot/dev releases.

The problem is that as of 0.6.11 `SerializationQueue::decode` can return `false` if it rejected the message, and the client isn't checking that: https://github.com/eclipse/milo/blob/7bd466ecae599f7651286e5d75709eb96edc0e08/opc-ua-stack/stack-client/src/main/java/org/eclipse/milo/opcua/stack/client/transport/uasc/UascClientMessageHandler.java#L617

Not that a check would do much good, because the only thing that can be done in that case it shut the channel down, and that ends up happening anyway.

The whole thing is non-deterministic and timing based.

ThoSap, 2024-05-02:
So I guess it would probably be best to throttle the async browse recursion using a FixedThreadPool Executor (limited by `MAX_CONCURRENT_BROWSE_TASKS`) + Semaphore in method BadSecurityCheckFailedExample#scrapeNodes, right?

kevinherron, 2024-05-02:
I am a bit curious why the maxQueueSize system property didn't work. Was it the exact same errors about sequence numbers that preceded other errors? Are you sure it took effect?

ThoSap, 2024-05-03:
> I am a bit curious why the maxQueueSize system property didn't work. Was it the exact same errors about sequence numbers that preceded other errors? Are you sure it took effect?

I just figured out why the system property did not work.
As I did not runtime initialize class `org.eclipse.milo.opcua.stack.core.channel.SerializationQueue` in the GraalVM AoT compiled native-image, the `private static final int MAX_QUEUE_SIZE = Integer.getInteger("milo.stack.serialization.maxQueueSize", 256);` was effectively inlined in the GraalVM native-image.

After configuring the class `org.eclipse.milo.opcua.stack.core.channel.SerializationQueue` to be runtime initialized by GraalVM native-image, I no longer get the exceptions above.

If you find it worthwhile you could add a check if this class was AoT compiled by GraalVM native-image without runtime initializing it using this approach:
https://github.com/FasterXML/jackson-databind/blob/jackson-databind-2.17.0/src/main/java/com/fasterxml/jackson/databind/util/NativeImageUtil.java
https://github.com/FasterXML/jackson-databind/blob/jackson-databind-2.17.0/src/main/java/com/fasterxml/jackson/databind/ser/impl/UnknownSerializer.java#L48-L51

The GraalVM technology is now part of the OpenJDK (called [Project Leyden](https://openjdk.org/projects/leyden/)), so this check could be important.
https://www.graalvm.org/2022/openjdk-announcement/

ThoSap, 2024-05-03:
To get Eclipse Milo to work with in a **GraalVM AoT compiled native-image** I added the following **runtime initialization** and **reflection** config:

**Runtime initialization**

The org.google.common.* classes are needed to be runtime initialized for Guava used in Eclipse Milo
```bash
--initialize-at-run-time=com.digitalpetri.netty.fsm.ChannelFsmConfigBuilder$SharedExecutor,com.google.common.cache.LongAddables,com.google.common.cache.Striped64,com.google.common.cache.Striped64$Cell,org.eclipse.milo.opcua.stack.core.channel.SerializationQueue,org.eclipse.milo.opcua.stack.core.util.BufferUtil,org.eclipse.milo.opcua.stack.core.util.NonceUtil
```
Also add the following classes for the server/client configuration to also print the Eclipse Milo and Java version in the `native-image` also add the following to the list above:
- Eclipse Milo client -> `org.eclipse.milo.opcua.sdk.client.OpcUaClient` 
- Eclipse Milo server -> `org.eclipse.milo.opcua.sdk.server.OpcUaServer,org.eclipse.milo.opcua.sdk.server.Session`

**Reflection**

To register necessary classes for reflection in the `native-image`, else the client won't work:
- For Quarkus:
  This class can be located in any package and can be named however you like:
```java
package com.example.graal;

import io.quarkus.runtime.annotations.RegisterForReflection;

/**
 * @author thomas.sapelza
 */
@RegisterForReflection(
    targets = {
        org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.AxisScaleEnumeration.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.AxisScaleEnumeration.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseDirection.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseDirection.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseResultMask.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.BrowseResultMask.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.DataChangeTrigger.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.DataChangeTrigger.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.DeadbandType.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.DeadbandType.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.ExceptionDeviationFormat.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.ExceptionDeviationFormat.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.FilterOperator.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.FilterOperator.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.HistoryUpdateType.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.HistoryUpdateType.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.IdType.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.IdType.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.MessageSecurityMode.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.MessageSecurityMode.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.ModelChangeStructureVerbMask.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.ModelChangeStructureVerbMask.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.MonitoringMode.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.MonitoringMode.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.NamingRuleType.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.NamingRuleType.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.NodeAttributesMask.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.NodeAttributesMask.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.NodeClass.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.NodeClass.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.OpenFileMode.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.OpenFileMode.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.PerformUpdateType.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.PerformUpdateType.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.RedundancySupport.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.RedundancySupport.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.SecurityTokenRequestType.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.SecurityTokenRequestType.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.ServerState.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.ServerState.Codec.class,
        org.eclipse.milo.opcua.stack.core.types.enumerated.StructureType.class,
[... 446 lines cut ...]
    "allDeclaredMethods": true,
    "allPublicMethods": true,
    "allDeclaredFields": true,
    "allPublicFields": true
  },
  {
    "name": "org.eclipse.milo.opcua.stack.core.types.enumerated.UserTokenType",
    "allDeclaredConstructors": true,
    "allPublicConstructors": true,
    "allDeclaredMethods": true,
    "allPublicMethods": true,
    "allDeclaredFields": true,
    "allPublicFields": true
  },
  {
    "name": "org.eclipse.milo.opcua.stack.core.types.enumerated.UserTokenType$Codec",
    "allDeclaredConstructors": true,
    "allPublicConstructors": true,
    "allDeclaredMethods": true,
    "allPublicMethods": true,
    "allDeclaredFields": true,
    "allPublicFields": true
  }
]
```

kevinherron, 2024-05-03:
Why do the enums to need get registered for anything to do with reflection?

ThoSap, 2024-05-03:
If I do not add these to the reflection config I will get the following error when connecting as a client.
This is probably due to how GraalVM optimizes the Bytecode to a Linux ELF native-image, see https://www.graalvm.org/latest/reference-manual/native-image/#native-image-and-third-party-libraries.

@ctron probably registered the same UaEnumeration classes here in this issue.
https://github.com/quarkusio/quarkus/issues/13114#issuecomment-1257630131
[image]

```java
__  ____  __  _____   ___  __ ____  ______
 --/ __ \/ / / / _ | / _ \/ //_/ / / / __/
 -/ /_/ / /_/ / __ |/ , _/ ,< / /_/ /\ \
--\___\_\____/_/ |_/_/|_/_/|_|\____/___/
2024-05-03T14:33:53.815Z INFO  [kafka] (smallrye-kafka-producer-thread-0) SRMSG18258: Kafka producer kafka-producer-message, connected to Kafka brokers 'kafka-bootstrap.example.com:443', is configured to write records to 'message'
2024-05-03T14:33:53.818Z INFO  [StartupOpcUaClient] (main) Eclipse Milo milo.stack.serialization.maxQueueSize: 2147483647
2024-05-03T14:33:53.818Z INFO  [StartupOpcUaClient] (main) Starting OPC UA client for address: opc.tcp://mykepwarevm.example.com:49320
2024-05-03T14:33:53.819Z INFO  [OpcUaClient] (main) Java version: 21.0.2
2024-05-03T14:33:53.819Z INFO  [OpcUaClient] (main) Eclipse Milo OPC UA Stack version: 0.6.12
2024-05-03T14:33:53.819Z INFO  [OpcUaClient] (main) Eclipse Milo OPC UA Client SDK version: 0.6.12
2024-05-03T14:33:53.885Z ERROR [UascClientMessageHandler] (milo-shared-thread-pool-0) Error decoding UaResponseMessage: org.eclipse.milo.opcua.stack.core.UaSerializationException: java.lang.NoSuchMethodException: org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType.from(int)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readEnum(OpcUaBinaryStreamDecoder.java:689)
        at org.eclipse.milo.opcua.stack.core.types.structured.ApplicationDescription$Codec.decode(ApplicationDescription.java:119)
        at org.eclipse.milo.opcua.stack.core.types.structured.ApplicationDescription$Codec.decode(ApplicationDescription.java:108)
        at org.eclipse.milo.opcua.stack.core.serialization.codecs.GenericDataTypeCodec$GenericBinaryDataTypeCodec.decode(GenericDataTypeCodec.java:47)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readStruct(OpcUaBinaryStreamDecoder.java:700)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.lambda$readStruct$0(OpcUaBinaryStreamDecoder.java:712)
        at java.base@21.0.2/java.util.Optional.map(Optional.java:260)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readStruct(OpcUaBinaryStreamDecoder.java:712)
        at org.eclipse.milo.opcua.stack.core.types.structured.EndpointDescription$Codec.decode(EndpointDescription.java:125)
        at org.eclipse.milo.opcua.stack.core.types.structured.EndpointDescription$Codec.decode(EndpointDescription.java:116)
        at org.eclipse.milo.opcua.stack.core.serialization.codecs.GenericDataTypeCodec$GenericBinaryDataTypeCodec.decode(GenericDataTypeCodec.java:47)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readStructArray(OpcUaBinaryStreamDecoder.java:1207)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readStructArray(OpcUaBinaryStreamDecoder.java:1222)
        at org.eclipse.milo.opcua.stack.core.types.structured.GetEndpointsResponse$Codec.decode(GetEndpointsResponse.java:78)
        at org.eclipse.milo.opcua.stack.core.types.structured.GetEndpointsResponse$Codec.decode(GetEndpointsResponse.java:69)
        at org.eclipse.milo.opcua.stack.core.serialization.codecs.GenericDataTypeCodec$GenericBinaryDataTypeCodec.decode(GenericDataTypeCodec.java:47)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readMessage(OpcUaBinaryStreamDecoder.java:669)
        at org.eclipse.milo.opcua.stack.client.transport.uasc.UascClientMessageHandler.lambda$onSecureMessage$14(UascClientMessageHandler.java:654)
        at org.eclipse.milo.opcua.stack.core.channel.SerializationQueue.lambda$decode$1(SerializationQueue.java:63)
        at org.eclipse.milo.opcua.stack.core.util.TaskQueue$TaskWrapper.run(TaskQueue.java:273)
        at java.base@21.0.2/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1144)
        at java.base@21.0.2/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:642)
        at java.base@21.0.2/java.lang.Thread.runWith(Thread.java:1596)
        at java.base@21.0.2/java.lang.Thread.run(Thread.java:1583)
        at org.graalvm.nativeimage.builder/com.oracle.svm.core.thread.PlatformThreads.threadStartRoutine(PlatformThreads.java:833)
        at org.graalvm.nativeimage.builder/com.oracle.svm.core.posix.thread.PosixPlatformThreads.pthreadStartRoutine(PosixPlatformThreads.java:211)
Caused by: java.lang.NoSuchMethodException: org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType.from(int)
        at java.base@21.0.2/java.lang.Class.checkMethod(DynamicHub.java:1075)
        at java.base@21.0.2/java.lang.Class.getDeclaredMethod(DynamicHub.java:1165)
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readEnum(OpcUaBinaryStreamDecoder.java:685)
        ... 25 more

[... cut ...]
        at org.eclipse.milo.opcua.stack.core.serialization.OpcUaBinaryStreamDecoder.readEnum(OpcUaBinaryStreamDecoder.java:685)
        ... 25 more

2024-05-03T14:33:53.951Z ERROR [StartupOpcUaClient] (main) Could not create the OPC UA client
java.util.concurrent.ExecutionException: org.eclipse.milo.opcua.stack.core.UaSerializationException: java.lang.NoSuchMethodException: org.eclipse.milo.opcua.stack.core.types.enumerated.ApplicationType.from(int)
2024-05-03T14:33:53.951Z INFO  [StartupOpcUaClient] (main) Waiting 5 seconds to make another connection attempt...
```

kevinherron, 2024-05-03:
Ah, because of this: https://github.com/eclipse/milo/blob/7bd466ecae599f7651286e5d75709eb96edc0e08/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/serialization/OpcUaBinaryStreamDecoder.java#L679-L691

I don't think this will be an issue in 1.0. The enum decoding, at least in this spot, no longer uses reflection:
https://github.com/eclipse/milo/blob/a79fe4613e346db08c2f3bef96f34c6b00e10420/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/encoding/binary/OpcUaBinaryDecoder.java#L689-L691

kevinherron, 2024-05-03:
Yes that pattern is pretty widely used. If there is some kind of Graal configuration file projects can provide that might be worth pursuing for the 1.0 release. I'm interested in supporting Graal native image, but not currently a user, so I'm not familiar with this and wouldn't able to verify anything.