eclipse-milo/milo #1272: Vulnerabilities CVE-2024-30172/CVE-2024-30171/CVE-2024-29857 for bouncycastle

emericcolombe, 2024-05-13:
Bouncycasle 1.75 on which we currently depend has several reported vulnerabilities:

- https://www.cve.org/CVERecord?id=CVE-2024-30172
- https://www.cve.org/CVERecord?id=CVE-2024-30171
- https://www.cve.org/CVERecord?id=CVE-2024-29857

These have all been fixed in 1.78

emericcolombe, 2024-05-13:
Thanks for the merge @kevinherron ! Do you think a 6.13 release could be done ?
I guess your netty upgrade a few weeks back was related to the same kind of issues.

kevinherron, 2024-05-13:
@emericcolombe I'll do a pass to see if there's any other dependency updates and then I should be able to cut a release this week.

kevinherron, 2024-05-18:
@emericcolombe 0.6.13 has been released.

emericcolombe, 2024-05-21:
Thanks a lot !