eclipse-milo/milo #1299: [Question] Reference implementation of server side X509IdentityValidator

devaskim, 2024-08-26:
@kevinherron  Is there any reference implementation, guide or at least some tips how to validate client certificate on a server side?

I looked at [client implementation of certificate identity validator](https://github.com/eclipse/milo/blob/master/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/api/identity/X509IdentityProvider.java), I found [X509IdentityValidator](https://github.com/eclipse/milo/blob/master/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/identity/X509IdentityValidator.java), but unfortunately I still don't understand on what server should rely on validating client certificate. 

At least now my server implementation is almost the same as [official Milo server example](https://github.com/eclipse/milo/tree/master/milo-examples/server-examples).

kevinherron, 2024-08-26:
The spec doesn't really go into what it means to validate an X509 certificate used as an identity because it's really up to the application/implementation, which is why `X509IdentityValidator` just accepts a `Predicate`.

What I've seen other implementations do is maintain a separate `user-pki` folder in parallel to the main `pki` folder that is used to manage application instance certificates. (this would be `DefaultCertificateManager`)

"Validating" the certificate is on you, but a good starting place would be to do the same cert path building and checking that the validators do: https://github.com/eclipse/milo/blob/74ce06cae0e1663e424152fd3456b90b7f45c14c/opc-ua-stack/stack-server/src/main/java/org/eclipse/milo/opcua/stack/server/security/DefaultServerCertificateValidator.java#L63-L120

What you don't want are the additional checks for application URI etc... because those are specific to application instance certificates.

Ultimately what it would mean for an identity certificate to be validated would be:
- it's self-signed, and explicitly trusted 
- it's ca-signed, and the CA is trusted

devaskim, 2024-08-26:
Thanks a lot for quick and detailed answer

closed 2024-08-26 by devaskim