eclipse-milo/milo #1320: [Feature] Testcontainers module

devaskim, 2024-09-16:
Is there any plan to post [Milo example server](https://github.com/eclipse/milo/tree/master/milo-examples/server-examples) as [Testcontainers module](https://testcontainers.com/modules/)? Would be very helpful to have it as fast-deployable test environment

kevinherron, 2024-09-16:
Nothing imminent, but it's a good idea.

It would probably be some containerized version of the [demo server](https://github.com/digitalpetri/opc-ua-demo-server), though, rather than the example server.

devaskim, 2024-09-16:
> It would probably be some containerized version of the [demo server](https://github.com/digitalpetri/opc-ua-demo-server)

Exactly! I use it for testing our client - very very helpful!!!
Do you need any help/contribution for preparing such module?

kevinherron, 2024-09-16:
There's nothing stopping you or anybody else from packaging up your own version of this right now.

I'm not going to look into this any further until after 1.0 of both Milo and the demo server has been released. I suspect the demo server may need some modifications to be more "container friendly" but I'm not certain.

devaskim, 2025-06-03:
Hi @kevinherron,
Do you have a time to look at this task again?

devaskim, 2025-06-03:
>  Is there anything else even necessary?

Sorry, don't know about ability to wrap demo server as docker container. In general, well enougth to write tests)
Just curios, any plans to publish container on Docker Hub or something similar?