eclipse-milo/milo #1325: [Question] Is there any listener in Server SDK to catch client disconnection?

devaskim, 2024-09-24:
Once a client connect I do some routine, specific for my app. So I want to clean up some resource, after client disconnects. Is there any API to catch up such events?

devaskim, 2024-10-03:
@kevinherron Any ides here?

kevinherron, 2024-10-03:
I think the only thing close to this functionality would be adding a `SessionListener` to the Server's`SessionManager`: https://github.com/eclipse/milo/blob/d73a2165b9b3cf9f9cb0dd8311c19e2adab03863/opc-ua-sdk/sdk-server/src/main/java/org/eclipse/milo/opcua/sdk/server/SessionManager.java#L130-L137

devaskim, 2024-10-03:
Looks very promising, @kevinherron Thanks for help

kevinherron, 2024-10-03:
This should be what matters for cleaning up resources anyway, because a _connection_ can be temporary and the nature of it could depend on the underlying transport (if support for additional transports materializes, anyway...). 

A client can connect, create a Session, temporarily disconnect, reconnect, and then resume that Session. It's not until the Session is closed, either explicitly by the client or because it times out, that you would probably want to clean up the resources.