eclipse-milo/milo #1366: Move modules that aren't ready for 1.0 to incubating repo

kevinherron, 2025-01-25:
Create a new repo to hold "incubating" modules that aren't mature enough to include in the 1.0 release.

Right now, this includes:
- `codec-json`
- `transport-https`
- `transport-websocket`

The quality of `codec-json` is good enough but I'm not sure I want to commit to the strictly using the JSON representation defined by the OPC UA JSON encoding. I think there are probably better ways to represent things like Unions and OptionSets, and even built-in types like ByteString and a few others if you're trying to use JSON to represent runtime/dynamic types in your application.

kevinherron, 2025-01-25:
Should probably add `encoding-xml` to incubating as well unless #166 gets implemented.

system24seven, 2025-03-23:
#166 is sitting in progress on my end, running out or time to work on it. Also there is a lot of unknowns about implementation due to lack of examples.

kevinherron, 2025-03-23:
It's a tricky one. I think there are two good sources beside the spec:
- Knowing that it has to be the dual of the decoder, which actually is implemented for the most part
- Pillaging UANodeSet files and looking at cases where variable instances have a `<Value>`, which will be XML encoded. 

I'll eventually get to it if you don't because I'll need it to generate the values in UANodeSet files myself.

system24seven, 2025-03-23:
From what I have seen, the decoder is pretty lazy and there are a bunch of complex data types missing. 
UANodeSet files are where I am finding most of the info I have used, but I feel like I have made some assumptions on incomplete data that will bite me later...
I have a branch in my repository that is my current progress so far.

kevinherron, 2025-03-23:
@system24seven I've actually got a UANodeSet export that was done against the new [demo server](https://github.com/digitalpetri/opc-ua-demo-server), and consequently should have nodes with values for every datatype in scalar, array, and matrix variations.

[TestMilo_2.xml.zip](https://github.com/user-attachments/files/19406599/TestMilo_2.xml.zip)

kevinherron, 2025-04-10:
I think I'm going to leave the `dtd-*` modules in and just mark their packages deprecated.

kevinherron, 2025-04-15:
Ok, this is done now.

Deprecated `dtd-*` modules have had all their packages marked as deprecated for removal. They will be removed in some future `2.0` version, and essentially un-maintained except to keep them compiling in `1.x`.

"Incubating" modules now have a `README.md` explaining their status, they have a fixed `0.1-SNAPSHOT` version, and that version isn't used anyway because they are all configured to be skipped during Maven `deploy`.

PR #1433 (2025-04-15): Add `README.md` to "incubating" modules
...and set the module versions to 0.1-SNAPSHOT.

fixes #1366
closed 2025-04-15 by kevinherron