eclipse-milo/milo #1375: Client support network fault listening.

greenpupil, 2025-02-17:
Current status:
Client-side can listen SessionActivity Status.

Problem:
1.When the network fault occurs, client-side cannot detect the specific network fault, and client side cannot record corresponding logs.
2.When client-side reconnection(trigger by netty-channel-fms) fails, the cause of the reconnection failure cannot be reported, and the client-side cannot record logs.

Requirement:
The SDK-client provides an interface to listen on network connection exceptions.

kevinherron, 2025-02-17:
In the `dev/1.0` branch you can get the underlying transport and add a listener to the ChannelFsm:
```java
OpcUaClient client = ...;

OpcClientTransport transport = client.getTransport();

if (transport instanceof OpcTcpClientTransport tcp) {
  ChannelFsm channelFsm = tcp.getChannelFsm();

  channelFsm.addTransitionListener(
      (State from, State to, Event via) ->
          System.out.printf("%s -> %s via %s%n", from, to, via));
}
```

some of the `Event` types have a `Throwable` associated with them.

See [State](https://github.com/digitalpetri/netty-channel-fsm/blob/master/src/main/java/com/digitalpetri/netty/fsm/State.java) and [Event](https://github.com/digitalpetri/netty-channel-fsm/blob/master/src/main/java/com/digitalpetri/netty/fsm/Event.java) source.

kevinherron, 2025-02-18:
womp womp.

I'll have to make changes in that library and then bump the dependency.

closed 2025-02-19 by greenpupil

kevinherron, 2025-04-18:
@greenpupil I had forgotten about this... until now. Just merged it, in time for upcoming `1.0.0` release 😅

PR #1439 (2025-04-18): Update netty-channel-fsm to version 1.0.1
fixes #1375
commit bfa0611b (2025-04-18): Update netty-channel-fsm to version 1.0.1 (#1439)

fixes #1375