eclipse-milo/milo #1419: No interface to add custom handler to netty channel pipline

BrightVan, 2025-04-10:
Can't Find It

kevinherron, 2025-04-10:
It's possible when building a Client in the 1.0 branch: https://github.com/eclipse-milo/milo/blob/06ea1341d2a02ea2f99743c53229f5a1bdd40f6b/opc-ua-stack/transport/src/main/java/org/eclipse/milo/opcua/stack/transport/client/tcp/OpcTcpClientTransportConfigBuilder.java#L89-L102

No equivalent exists for the Server.

What kind of customization are you trying to do?

BrightVan, 2025-04-10:
> It's possible when building a Client in the 1.0 branch:
> 
> [milo/opc-ua-stack/transport/src/main/java/org/eclipse/milo/opcua/stack/transport/client/tcp/OpcTcpClientTransportConfigBuilder.java](https://github.com/eclipse-milo/milo/blob/06ea1341d2a02ea2f99743c53229f5a1bdd40f6b/opc-ua-stack/transport/src/main/java/org/eclipse/milo/opcua/stack/transport/client/tcp/OpcTcpClientTransportConfigBuilder.java#L89-L102)
> 
> Lines 89 to 102 in [06ea134](/eclipse-milo/milo/commit/06ea1341d2a02ea2f99743c53229f5a1bdd40f6b)
> 
>    /** 
>     * Set a {@link Consumer} that will be given a chance to customize the {@link ChannelPipeline} 
>     * used by this transport. 
>     * 
>     * @param channelPipelineCustomizer a {@link Consumer} that will be given a chance to customize 
>     *     the {@link ChannelPipeline} used by this transport. 
>     * @return this {@link OpcTcpClientTransportConfigBuilder}. 
>     */ 
>    public OpcTcpClientTransportConfigBuilder setChannelPipelineCustomizer( 
>        Consumer<ChannelPipeline> channelPipelineCustomizer) { 
>   
>      this.channelPipelineCustomizer = channelPipelineCustomizer; 
>      return this; 
>    } 
> No equivalent exists for the Server.
> 
> What kind of customization are you trying to do?
I need  equivalent exists for the Server, to check client address is in white list. so i want to add a channel inboundHandler to pipeline

BrightVan, 2025-04-10:
I need  equivalent exists for the Server, to check client address is in white list. so i want to add a channel inboundHandler to pipeline

BrightVan, 2025-04-10:
we can do like this
1. define a interface
```
public interface UaStackChannelInitInvoker {
    void initChannel(Channel channel);
}
```
2. add a function in `UaStackServerConfig`
```
    UaStackChannelInitInvoker getChannelInitInvoker();
```
3. add to UaStackServerConfigBuilder
```
    public UaStackServerConfigBuilder setChannelInitInvoker(UaStackChannelInitInvoker channelInitInvoker) {
        this.channelInitInvoker = channelInitInvoker;
        return this;
    }
```
4. add to OpcUaServerConfigBuilder 
```
    @Override
    public OpcUaServerConfigBuilder setChannelInitInvoker(UaStackChannelInitInvoker channelInitInvoker) {
        super.setChannelInitInvoker(channelInitInvoker);
        return this;
    }
```
5. use it 
```
 OpcUaServerConfig serverConfig = OpcUaServerConfig.builder()
            .setProductUri("urn:eclipse:milo:example-server")
            .setChannelInitInvoker(channel -> channel.pipeline().addLast("ChannelConnectHandler", new ChannelConnectHandler()))
            .build();
```

kevinherron, 2025-04-10:
Fixed in #1420 for 1.0.

closed 2025-04-10 by kevinherron

BrightVan, 2025-04-11:
so when will version 1.0 be released

kevinherron, 2025-04-11:
Maybe on May 1. Need to figure out the bureaucracy around doing a 1.0.0 release with the Eclipse Foundation. 

It should be pretty stable until then. You may as well start your migration now.

BrightVan, 2025-04-11:
ok , thanks for your good work.

PR #1420 (2025-04-10): Allow customization of ServerBoostrap and ChannelPipeline
fixes #1419
commit 8208aafb (2025-04-10): Allow customization of ServerBoostrap and ChannelPipeline (#1420)

fixes #1419