eclipse-milo/milo #1465: Request for Migration Guide to Upgrade to Version 1.0.0

rangoy, 2025-05-20:
I would like to request a migration guide to help users upgrade to version 1.0.0. Please include:
- Key breaking changes from previous versions
- Step-by-step instructions for upgrading
- Example code or configuration updates
- Any known issues or caveats

A comprehensive guide will assist users in making a smooth transition to the new version.

For reference, the Apache Camel team (see https://issues.apache.org/jira/browse/CAMEL-22086) has indicated they need a migration guide to begin upgrading the Camel Milo component. Providing such a guide will help downstream projects like Camel smoothly adopt the new version.

peterfranciscook, 2025-08-05:
Sharing notes I've taken on my upgrade journey. Successfully packaged app locally with mvn after these steps, afraid to touch bazel build until I figure a couple more TODOs out. I will update as a figure more things out. 

(before all else) forked main branch / created new branch `tech/milo-1-x-update`
# Update dependencies / build files
## maven dependency updates
### Update pom.xml
#### Before:
```xml
        <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>sdk-client</artifactId>
            <version>0.6.16</version>
        </dependency>

        <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>stack-core</artifactId>
            <version>0.6.16</version>
        </dependency>
		
```
#### After:
```xml
        <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>milo-sdk-client</artifactId>
            <version>1.0.2</version>
        </dependency>

        <dependency>
            <groupId>org.eclipse.milo</groupId>
            <artifactId>milo-stack-core</artifactId>
            <version>1.0.2</version>
        </dependency>
```
### Clean/Refresh IDE Workspace(VSCode)
1. hit the F1 key, type "java clean" and run the "Clean Java Server Workspace" command
2. Close the IDE and run `mvn clean` in bash just to be safe
## bazel dependency updates
1. Update Maven Packages in MODULE.bazel
```starlark
# before
"org.eclipse.milo:sdk-client:0.6.16",
"org.eclipse.milo:stack-core:0.6.16",
"org.eclipse.milo:stack-client:0.6.16",
# after
"org.eclipse.milo:milo-sdk-client:1.0.2",
"org.eclipse.milo:milo-stack-core:1.0.2",
# NOTE: stack-client appears to have gone away
# TODO: bazel/buildozer will probably howl about additional transitive dependencies
```
2. Update all your module `BUILD` files to reference the new artifacts
3. Close the IDE and run `bazel clean --expunge` in bash

## Update milo packages in your code
With my IDE still closed, I started by running `mvn package -DskipTests | 'grep symbol:'` to find my initial list of broken dependencies:
```
[ERROR]   symbol: class ValueConsumer
[ERROR]   symbol: class DataTypeTreeSessionInitializer
[ERROR]   symbol: class UaMonitoredItem
[ERROR]   symbol: class DataTypeTree
[ERROR]   symbol: class UaStackClient
[ERROR]   symbol: class UaSubscription
[ERROR]   symbol: class OpcUaBinaryDataTypeCodec
[ERROR]   symbol: class UaStructure
[ERROR]   symbol: class GenericDataTypeCodec
[ERROR]   symbol: class SerializationContext
[ERROR]   symbol: class UaDecoder
[ERROR]   symbol: class UaEncoder
```
[... 82 lines cut ...]
import org.eclipse.milo.opcua.stack.core.encoding.GenericDataTypeCodec;
```
```java
// find and replace this
import org.eclipse.milo.opcua.stack.core.serialization.SerializationContext;
// ... with this
import org.eclipse.milo.opcua.stack.core.encoding.EncodingContext;
```
- repeat for class reference `SerializationContext` -> `EncodingContext`
```java
// find and replace these
import org.eclipse.milo.opcua.stack.core.serialization.UaDecoder;
import org.eclipse.milo.opcua.stack.core.serialization.UaEncoder;
// ... with this
import org.eclipse.milo.opcua.stack.core.encoding.UaDecoder;
import org.eclipse.milo.opcua.stack.core.encoding.UaEncoder;
```
- find and replace UaDecoder calls like this `decoder.read[Type]` with `decoder.decode[Type]`
- find and replace UaEncoder calls like this `encoder.write[Type]` with `encoder.encode[Type]`
- find and replace Codec inner class override methods `decode` -> `decodeType` & `encode` -> `encodeType`
- ??? find and replace `OpcUaBinaryDataTypeCodec` -> `GenericDataTypeCodec` ??? TODO: verify this works for registering codecs at startup ??? TODO maybe use interface `DataTypeCodec` instead ???
  - also replace any calls to `[T].Codec().asBinaryCodec()` to just `[T].Codec()`

## DataTypeTreeSessionInitializer & DataTypeTree
These classes have gone away, mostly I was using these in debug stacks and unit tests to inspect codec registration failures. Their replacement is unclear

peterfranciscook, 2025-08-07:
## ExtensionObject vs DynamicStuctType
In milo 0.6.x calling `OpcUaClient::readValues` would return a CompletableFuture<List<DataValue>>`, I would fetch the `ExtensionObject` to decode by calling `.getValue().getValue()`.

The (roughly) similar call chain in milo 1.x yields `DynamicStructType` (not what I'm looking for). 

After looking at milo examples I realized I needed to change my codec registration to use the `DataTypeManager` from `OpcUaClient::getStaticDataTypeManager` instead of `OpcUaClient::getDynamicDataTypeManager`, and decode using the `EncodingContext` from `OpcUaClient::getStaticEncodingContext` rather than `OpcUaClient::getDynamicEncodingContext`

kevinherron, 2025-08-08:
@peterfranciscook can you detail this a little more?

> The async method (OpcUaClient::writeValuesAsync) in milo 1.x will quietly fail and does not return a failed future if you attempt this.

I can't reproduce:

```
  @Test
  void asyncWriteWithNullStatusCodeDoesNotHang() throws Exception {
    NodeId nodeId = testNodeId();

    //noinspection DataFlowIssue
    DataValue dvWithNullStatus =
        DataValue.newValue().setValue(new Variant(123)).setStatus(null).build();

    CompletableFuture<List<StatusCode>> future =
        client.writeValuesAsync(List.of(nodeId), List.of(dvWithNullStatus));

    final CountDownLatch latch = new CountDownLatch(1);
    final AtomicBoolean completedExceptionally = new AtomicBoolean(false);
    final AtomicBoolean completedSuccessfully = new AtomicBoolean(false);

    future.whenComplete(
        (sc, ex) -> {
          if (ex != null) {
            ex.printStackTrace(System.err);
            completedExceptionally.set(true);
          } else {
            System.out.println("Completed: " + sc);
            completedSuccessfully.set(true);
          }
          latch.countDown();
        });

    assertTrue(latch.await(2, TimeUnit.SECONDS), "expected async writeValues() to complete");
    assertTrue(
        completedExceptionally.get(), "expected async writeValues() to complete exceptionally");
    assertFalse(
        completedSuccessfully.get(), "expected async writeValues() to complete exceptionally");
  }
```

peterfranciscook, 2025-08-11:
> [@peterfranciscook](https://github.com/peterfranciscook) so what's the issue? Your test demonstrates the same thing mine does - the CompletableFuture _does_ complete (exceptionally), with the NPE reported in the nested exceptions.
> 
> It _is_ expected that forcing a null StatusCode in the DataValue results in an NPE now. But this isn't failing silently.

Ohhhhh ok, I was not understanding you on Friday. 

When I say It "fails silently" in an async context, I meant this - if I call `writeValuesAsync` (passed a `DataValue` with null statusCode), with a `.whenComplete` callback, but without a later call to `join` or `get`, it won't complete (normally or exceptionally) on its own.