eclipse-milo/milo #1527: when i use sign i got the trustAnchors parameter must be non-empty

czndtb, 2025-07-30:
```java
Path securityTempDir = Paths.get(System.getProperty("java.io.tmpdir"), "client", "security");
        Files.createDirectories(securityTempDir);
        if (!Files.exists(securityTempDir)) {
            throw new Exception("unable to create security dir: " + securityTempDir);
        }

        Path pkiDir = securityTempDir.resolve("pki");

        log.info("security dir: {}", securityTempDir.toAbsolutePath());
        log.info("security pki dir: {}", pkiDir.toAbsolutePath());

        KeyStoreLoader loader = new KeyStoreLoader().load(securityTempDir);

        clientTrustListManager = FileBasedTrustListManager.createAndInitialize(pkiDir);

        // 默认匿名
        IdentityProvider identityProvider;

        if (configuration.getUsername() != null && configuration.getPassword() != null) {
            identityProvider = new UsernameProvider(configuration.getUsername(), configuration.getPassword());
        } else {
            identityProvider = new AnonymousProvider();
        }

        var certificateValidator =
                new DefaultClientCertificateValidator(
                        clientTrustListManager, new MemoryCertificateQuarantine());

        return OpcUaClient.create(
                configuration.getEndPointUrl(),
                endpoints -> {

                    return endpoints.stream()
                            .filter(e -> e.getSecurityPolicyUri().equals(configuration.getPolicy()))
                            .filter(e -> Arrays.stream(e.getUserIdentityTokens()).anyMatch(t -> t.getTokenType() == UserTokenType.UserName))
                            .findFirst();
//                    return endpoints.stream()
//                            .filter(e -> e.getSecurityPolicyUri().equals(configuration.getPolicy()))
//                            .findFirst();
                },
                transportConfigBuilder -> {
                },
                clientConfigBuilder ->
                        clientConfigBuilder
                                .setApplicationName(LocalizedText.english("eclipse milo opc-ua client"))
                                .setApplicationUri("urn:eclipse:milo:examples:client")
                                .setKeyPair(loader.getClientKeyPair())
                                .setCertificate(loader.getClientCertificate())
                                .setCertificateChain(loader.getClientCertificateChain())
                                .setCertificateValidator(certificateValidator)
                                .setIdentityProvider(identityProvider));
```

I use this code by got
```
io.netty.handler.codec.DecoderException: UaException: status=Bad_SecurityChecksFailed, message=java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:500)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:290)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1357)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:440)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:868)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:796)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:732)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:998)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.base/java.lang.Thread.run(Thread.java:840)
Caused by: org.eclipse.milo.opcua.stack.core.UaException: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at org.eclipse.milo.opcua.stack.core.util.validation.CertificateValidationUtil.buildCertPath(CertificateValidationUtil.java:388)
	at org.eclipse.milo.opcua.stack.core.util.validation.CertificateValidationUtil.buildTrustedCertPath(CertificateValidationUtil.java:132)
	at org.eclipse.milo.opcua.stack.core.security.DefaultClientCertificateValidator.validateCertificateChain(DefaultClientCertificateValidator.java:72)
	at org.eclipse.milo.opcua.stack.transport.client.uasc.UascClientMessageHandler.onOpenSecureChannel(UascClientMessageHandler.java:383)
	at org.eclipse.milo.opcua.stack.transport.client.uasc.UascClientMessageHandler.decode(UascClientMessageHandler.java:269)
	at io.netty.handler.codec.ByteToMessageCodec$1.decode(ByteToMessageCodec.java:42)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:530)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:469)
	... 17 common frames omitted
Caused by: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at java.base/java.security.cert.PKIXParameters.setTrustAnchors(PKIXParameters.java:200)
	at java.base/java.security.cert.PKIXParameters.<init>(PKIXParameters.java:120)
	at java.base/java.security.cert.PKIXBuilderParameters.<init>(PKIXBuilderParameters.java:104)
	at org.eclipse.milo.opcua.stack.core.util.validation.CertificateValidationUtil.buildCertPath(CertificateValidationUtil.java:357)
	... 24 common frames omitted
```

czndtb, 2025-07-31:
Thank you. This is effective

closed 2025-07-31 by czndtb

mlohbihler, 2026-02-03:
Note that this appears to be the case for the demo server at `opc.tcp://milo.digitalpetri.com:62541/milo`. I was getting the same error, but was able to write the server cert to the local trusted dir and it then worked.