eclipse-milo/milo #1591: Please Improve Exception traces/ UaException.extract

tg-freigmbh, 2025-09-17:
**Describe the bug**
I am currently investigating what is probably a a bug in our testsetup. For example I get an Exception like below.
However these stacktraces do not contain any information what was called from where.  This can be reproduced for example by not having a Server running and trying to connect using `OpcUaClient.create ` ->  since there is a line like
` throw UaException.extract(e).orElseGet(() -> new UaException(e));` you end up with the stacktrace below which means you wont know which method actually threw the Exception or who was calling it.

```
io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: getsockopt: /127.0.0.1:1443
UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: getsockopt: /127.0.0.1:1443
	at org.eclipse.milo.opcua.stack.transport.client.tcp.OpcTcpClientTransport$ClientChannelActions.lambda$connect$0(OpcTcpClientTransport.java:182)
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:603)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners0(DefaultPromise.java:596)
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:572)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:505)
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:649)
	at io.netty.util.concurrent.DefaultPromise.setFailure0(DefaultPromise.java:642)
	at io.netty.util.concurrent.DefaultPromise.tryFailure(DefaultPromise.java:131)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.fulfillConnectPromise(AbstractNioChannel.java:326)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:342)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:784)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:732)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:998)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.base/java.lang.Thread.run(Thread.java:1583)
Caused by: io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: getsockopt: /127.0.0.1:1443
Caused by: java.net.ConnectException: Connection refused: getsockopt
	at java.base/sun.nio.ch.Net.pollConnect(Native Method)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:336)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:339)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:784)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:732)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:998)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.base/java.lang.Thread.run(Thread.java:1583)

Connection refused: getsockopt: /127.0.0.1:1443
io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: getsockopt: /127.0.0.1:1443
Caused by: java.net.ConnectException: Connection refused: getsockopt
	at java.base/sun.nio.ch.Net.pollConnect(Native Method)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:336)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:339)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:784)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:732)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:998)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.base/java.lang.Thread.run(Thread.java:1583)

```

**Expected behavior**
Stacktraces containing information about which method was invoked how from usercode should never be removed

**Logs and Packet Captures**

**Additional context**
1.0.6

kevinherron, 2025-09-17:
When I try your example of using `OpcUaClient.create` against an offline host, I get this stack trace:
```
UaException: status=Bad_InternalError, message=java.util.concurrent.ExecutionException: io.netty.channel.AbstractChannel$AnnotatedSocketException: Host is down: /10.211.55.3:4842
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.lambda$create$6(OpcUaClient.java:302)
	at java.base/java.util.Optional.orElseGet(Optional.java:364)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.create(OpcUaClient.java:302)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.create(OpcUaClient.java:300)
	at Scratch.main(scratch.java:14)
Caused by: java.util.concurrent.ExecutionException: io.netty.channel.AbstractChannel$AnnotatedSocketException: Host is down: /10.211.55.3:4842
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:396)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2073)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.create(OpcUaClient.java:265)
	... 2 more
Caused by: io.netty.channel.AbstractChannel$AnnotatedSocketException: Host is down: /10.211.55.3:4842
Caused by: java.net.SocketException: Host is down
	at java.base/sun.nio.ch.Net.connect0(Native Method)
	at java.base/sun.nio.ch.Net.connect(Net.java:579)
	at java.base/sun.nio.ch.Net.connect(Net.java:586)
	at java.base/sun.nio.ch.SocketChannelImpl.connect(SocketChannelImpl.java:853)
	at io.netty.util.internal.SocketUtils$3.run(SocketUtils.java:91)
	at io.netty.util.internal.SocketUtils$3.run(SocketUtils.java:88)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:569)
	at io.netty.util.internal.SocketUtils.connect(SocketUtils.java:88)
	at io.netty.channel.socket.nio.NioSocketChannel.doConnect(NioSocketChannel.java:321)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.connect(AbstractNioChannel.java:250)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.connect(DefaultChannelPipeline.java:1289)
	at io.netty.channel.AbstractChannelHandlerContext.invokeConnect(AbstractChannelHandlerContext.java:655)
	at io.netty.channel.AbstractChannelHandlerContext.connect(AbstractChannelHandlerContext.java:634)
	at io.netty.channel.AbstractChannelHandlerContext.connect(AbstractChannelHandlerContext.java:618)
	at io.netty.channel.DefaultChannelPipeline.connect(DefaultChannelPipeline.java:927)
	at io.netty.channel.AbstractChannel.connect(AbstractChannel.java:264)
	at io.netty.bootstrap.Bootstrap$3.run(Bootstrap.java:264)
	at io.netty.util.concurrent.AbstractEventExecutor.runTask(AbstractEventExecutor.java:173)
	at io.netty.util.concurrent.AbstractEventExecutor.safeExecute(AbstractEventExecutor.java:166)
	at io.netty.util.concurrent.SingleThreadEventExecutor.runAllTasks(SingleThreadEventExecutor.java:472)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:569)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:998)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.base/java.lang.Thread.run(Thread.java:840)
```

The call site is the first exception in the chain:
```
UaException: status=Bad_InternalError, message=java.util.concurrent.ExecutionException: io.netty.channel.AbstractChannel$AnnotatedSocketException: Host is down: /10.211.55.3:4842
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.lambda$create$6(OpcUaClient.java:302)
	at java.base/java.util.Optional.orElseGet(Optional.java:364)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.create(OpcUaClient.java:302)
	at org.eclipse.milo.opcua.sdk.client.OpcUaClient.create(OpcUaClient.java:300)
	at Scratch.main(scratch.java:14)
```

kevinherron, 2025-09-17:
Ahh, it's different if underlying Netty error is different:
```
UaException: status=Bad_ConnectionRejected, message=io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: localhost/127.0.0.1:4840
	at org.eclipse.milo.opcua.stack.transport.client.tcp.OpcTcpClientTransport$ClientChannelActions.lambda$connect$0(OpcTcpClientTransport.java:182)
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:603)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners0(DefaultPromise.java:596)
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:572)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:505)
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:649)
	at io.netty.util.concurrent.DefaultPromise.setFailure0(DefaultPromise.java:642)
	at io.netty.util.concurrent.DefaultPromise.tryFailure(DefaultPromise.java:131)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.fulfillConnectPromise(AbstractNioChannel.java:326)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:342)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:784)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:732)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:998)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at java.base/java.lang.Thread.run(Thread.java:840)
Caused by: io.netty.channel.AbstractChannel$AnnotatedConnectException: Connection refused: localhost/127.0.0.1:4840
Caused by: java.net.ConnectException: Connection refused
	at java.base/sun.nio.ch.Net.pollConnect(Native Method)
	at java.base/sun.nio.ch.Net.pollConnectNow(Net.java:672)
	at java.base/sun.nio.ch.SocketChannelImpl.finishConnect(SocketChannelImpl.java:946)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:336)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:339)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:784)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:732)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:658)
```

kevinherron, 2025-09-22:
Other than touching a lot of code this doesn't look too bad. The `UaException(Throwable)` constructor becomes:

```java
  /**
   * Creates a new UaException with a cause.
   *
   * <p>The StatusCode is extracted from the cause or any of its nested causes by looking for an
   * implementation of {@link UaExceptionStatus}, otherwise {@link StatusCodes#Bad_UnexpectedError}
   * is used.
   *
   * @param cause the cause of this exception.
   */
  public UaException(Throwable cause) {
    super(cause);

    this.statusCode =
        UaExceptionStatus.extract(cause)
            .map(UaExceptionStatus::getStatusCode)
            .orElse(new StatusCode(StatusCodes.Bad_UnexpectedError));
  }
```

Then it's just a big search/replace effort to replace this pattern:
```java
throw UaException.extract(e).orElse(new UaException(StatusCodes.Bad_UnexpectedError, e));
```

with this:
```java
throw new UaException(e);
```

commit 09f6477f (2025-10-01): Preserve stack traces when throwing UaException (#1592)

Automatically extract StatusCode from nested exception causes in
UaException(Throwable) constructor by checking for UaExceptionStatus
implementations. Use this constructor to preserve stack traces instead
of throwing extracted exceptions. Standardize exception handling order
for ExecutionException and InterruptedException across the codebase.

fixes #1591
closed 2025-10-01 by kevinherron